igorauad/gr-dvbs2rx #2: Can't install to a custom prefix with sudo.

drmpeg, 2021-12-18:
When ```sudo make install``` tries to compile the heir block, the build fails. That's because the users environment (and PYTHONPATH) is lost when ```sudo``` is used.

igorauad, 2022-01-05:
Hi @drmpeg ,

Thanks for opening the issue and apologies for the delay. I still haven't investigated this issue because I need to set up an environment to reproduce it. Just so I understand:

- Are you trying to install as root with `sudo make install` after running `make` to build with the non-root user? Or did you try running `sudo make install` directly (with no preceding `make`) to execute both the build and the installation with the root user?

- Are you using a GR installation on a prefix that requires explicitly setting the PYTHONPATH? 

In other words, how can I reproduce the problem? I'm thinking I need to build GR from source and install it on a non-conventional prefix (i.e., out of `/usr/` and `/usr/local`). Then, I need to try building gr-dvbs2rx using this GR installation (on the non-conventional prefix). Is that right?

Also, do you think there is any easy workaround to be added at [the corresponding CMakeLists.txt](https://github.com/igorauad/gr-dvbs2rx/blob/master/grc/hier/CMakeLists.txt)?

I could also make the hierarchical block optional on CMake. It's only used by the example flowgraphs, and the flowgraphs are not built automatically by CMake, so building the hier block is only a convenience. The hierarchical block is not used by the `dvbs2-rx` application. Instead, this application connects the individual components (see https://github.com/igorauad/gr-dvbs2rx/blob/master/apps/dvbs2-rx#L606).

drmpeg, 2022-01-11:
You could just use the old method of generating hierarchical blocks. Put ```gr-dvbs2rx/grc/hier/dvbs2rx_rx_hier.grc``` in ```gr-dvbs2rx/examples```. When you load ```dvbs2_rx.grc```, the hierarchical block will be generated. Then just click the "Reload Blocks" button.

777arc, 2022-08-12:
Thank you so much @drmpeg, this is exactly what was tripping me up, I kept getting the error  `ImportError: generic_type: type "bbdeheader_bb" is already registered!` until I tried your suggestion, I also had to wipe ~/.grc_gnuradio (and I wiped the python portion of the install just to be safe) and reinstalled the OOT, and it finally worked!!!  Would highly recommend implementing this change, since so many people are going to be installing it to the system with sudo, especially now that the GR PPA is pretty much the go-to method for installing GR, vs pybombs.  Let me know if you want me to do the PR.

closed 2022-08-22 by igorauad

igorauad, 2022-08-22:
Hi, @777arc . I've implemented the change in a69b5947a9647f5623702d115bd570bd3fb7eb31 as suggested by @drmpeg . Thanks!

commit 2ac046fb (2022-08-18): Move Rx hier block to the examples directory

The Rx hier block is currently used by the examples only, and not by the
main dvbs2-tx/rx apps. Instead, the main apps connect the subblocks one
by one to allow for various customizations. As a result, the hier block
is currently optional just like the example flowgraphs, which means the
cmake build doesn't need to generate the hier block automatically.

By moving the hier block to the examples/ directory, any flowgraph using
it will trigger its generation when compiled. With that, we avoid
installation problems like the one reported in Issue #2.

 [... cut]
commit 93951765 (2022-08-20): Move Rx hier block to the examples directory

The Rx hier block is currently used by the examples only, and not by the
main dvbs2-tx/rx apps. Instead, the main apps connect the subblocks one
by one to allow for various customizations. As a result, the hier block
is currently optional just like the example flowgraphs, which means the
cmake build doesn't need to generate the hier block automatically.

By moving the hier block to the examples/ directory, any flowgraph using
it will trigger its generation when compiled. With that, we avoid
installation problems like the one reported in Issue #2.

 [... cut]
commit a69b5947 (2022-08-20): Move Rx hier block to the examples directory

The Rx hier block is currently used by the examples only, and not by the
main dvbs2-tx/rx apps. Instead, the main apps connect the subblocks one
by one to allow for various customizations. As a result, the hier block
is currently optional just like the example flowgraphs, which means the
cmake build doesn't need to generate the hier block automatically.

By moving the hier block to the examples/ directory, any flowgraph using
it will trigger its generation when compiled. With that, we avoid
installation problems like the one reported in Issue #2.

 [... cut]