igorauad/gr-dvbs2rx #3: Fails to build due to illegal conversion between _Complex double and float

marcusmueller, 2021-12-19:
using F34, cmake 3.20.5, clang 12.0.1 to build 878c4f77ae78c78c0aa9be753d5485a8b805a0ec
```
FAILED: lib/CMakeFiles/gnuradio-dvbs2rx.dir/pl_freq_sync.cc.o
ccache /usr/bin/clang++ -DGR_MPLIB_GMP -Dgnuradio_dvbs2rx_EXPORTS -I../lib/../include -isystem /home/marcus/usrlocal-3.9/include -isystem /usr/include/python3.9 -march=native -Wall -O2 -g -DNDEBUG -fPIC   -fvisibility=hidden -std=gnu++17 -MD -MT lib/CMakeFiles/gnuradio-dvbs2rx.dir/pl_freq_sync.cc.o -MF lib/CMakeFiles/gnuradio-dvbs2rx.dir/pl_freq_sync.cc.o.d -o lib/CMakeFiles/gnuradio-dvbs2rx.dir/pl_freq_sync.cc.o -c ../lib/pl_freq_sync.cc
../lib/pl_freq_sync.cc:88:37: error: implicit conversion from '_Complex double' to 'float' is not permitted in C++
        unmod_pilots[i] = (+SQRT2_2 - SQRT2_2i);
                        ~  ~~~~~~~~~^~~~~~~~~~
1 error generated.
```

marcusmueller, 2021-12-19:
everywhere like here, where it's clear the target type is `gr_complex==std::complex<float>`, you could do things like

    unmod_pilots[i] = {+SQRT2_2, -SQRT2_2};

instead, to call the `std::complex` constructor via initialization list. 

But this is done in a lot of places, and I'm not sure that trick works everywhere.

igorauad, 2022-01-05:
Your solution in 8ee89fda75c869553777c9db384a9e36ca42e380 has been merged to master already. Many thanks!

closed 2022-01-05 by igorauad