igorauad/gr-dvbs2rx #9: Building 3.10

razed11, 2022-04-20:
I decided to forego the Docker image and build on Ubuntu 20.04. The latest version for Linux is `3.9.5` in terms of using `apt-get`. Is there a way to build against this or will I have to build GNU Radio from source to get `3.10`?

```
$ cmake ..
-- Build type not specified: defaulting to release.
CMake Error at CMakeLists.txt:77 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.10".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.9.5.0
    /lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake, version: 3.9.5.0
```

igorauad, 2022-04-20:
Yes, you can use the `maint-3.9` branch if you'd like to run on GR 3.9. I've upgraded the master branch to GR 3.10 recently.

I think the error you are seeing might be due to not cleaning up the build directory. It seems like it still has some leftover from the master branch, which requires GR 3.10. The `maint-3.9` branch requires GR 3.9.2.  You can try rebuilding from an empty build directory.

razed11, 2022-04-22:
Sorry. I should have known to clean up that directory.

On Ubuntu 20.04 I had to `apt install libsndfile1-dev` and it then built correctly.

closed 2022-04-22 by razed11