igorauad/gr-dvbs2rx #10: QO100 beacon

sv1bds, 2022-05-31:
Hi,
I try to use it to receive QO100 beacon (Es Heil 2) without success until now.
I can get a MER of about 10.2  using Polyphase Clock Sync & Costas Loop.
But no constellation is archived after DVBS2 PL Sync.
I suspect that I have to set ACM/VCM mode to False but I do not know what values must entered at PLS filter entries.

Any help is welcome !!!

sv1bds, 2022-05-31:
[DVBS2RX.PDF](https://github.com/igorauad/gr-dvbs2rx/files/8805766/DVBS2RX.PDF)
[image]
A simplified block is attached and the result screen.
Thanks

igorauad, 2022-06-01:
Hi @sv1bds ,

Thanks for your interest in the project. I'd recommend using the `dvbs2-rx` app instead of the hierarchical block, given the app has many options. I assume there is a way to pipe the PlutoSDR IQ output into `dvbs2-rx`, so you would run it like so:

```
pluto-sdr-source | dvbs2-rx --sink file --out-file /dev/null --gui --sym-rate 1.5M --samp-rate 3M --modcod qpsk4/5 --rolloff 0.25 --pilots auto
```

In this example, `pluto-sdr-source` is a hypothetical application feeding IQ samples into `dvbs2-rx`. If you don't have one, you can create one on GRC. You can feed the output of the PlutoSDR Source block directly into a File Descriptor Sink block configured for file descriptor 1 (stdout) and a complex input type. Then, generate the flowgraph and run it while piping the output IQ into `dvbs2-rx`.

I have never tried receiving this DATV beacon myself, so I'm not sure if the PLFRAMEs contain pilot symbols and if the FECFRAMEs are of short or normal size. I saw on your screenshot that you are configuring for short FECFRAMEs. If that's the case, you should add `--frame-size short` to the above command. Also, if you are sure there are no pilots, you can change from `--pilots auto` to `--pilots off`.

Alternatively, it is often helpful to run the low-level PHY block in ACM/VCM mode to double-check the unknown settings like FECFRAME size and the presence of pilots. To do so, run like so:

```
pluto-sdr-source | dvbs2-rx --sink file --out-file /dev/null --gui --sym-rate 1.5M --samp-rate 3M --modcod qpsk4/5 --rolloff 0.25 --pl-acm-vcm -d1
```

Option `--pl-acm-vcm` applies the ACM/VCM setting at low-level, while option `-d1` sets the first debugging level (chose a higher number for even more logs). Then, if everything goes well, you might see lines like:

```
plsc_decoder :debug: MODCOD: 8; Short FECFRAME: 0; Pilots: 0
```

Based on that, you will know the correct settings. For example, the above would mean QPSK 4/5 with normal FECFRAME and no pilots. The MODCOD number you can see in Table 12 on the standard.

After that, you can go back to the regular command (without the `--pl-acm-vcm` flag) and just provide the exact configuration to it.

Let me know if that helps. And if you have an IQ recording to share, I'd be happy to try processing it.

Also, please note I have just pushed a couple of commits with some fixes and improvements. I recommend you pull them and try with the most up-to-date version. Besides, there was a fix there for the ACM/VCM mode.

sv1bds, 2022-06-02:
Hi Igor,
Thanks for your valuable information!
To update code I just overide it?
I prefer to write my own code as I write Python in my job.
Your code is very useful for an end user.
Thanks a lot for your effort!!!
Any documention of using blocks will be appreciated.
 George SV1BDS

igorauad, 2022-07-05:
Hi @sv1bds , apologies for the delay. I'm catching up with some pending issues now. Were you able to figure out how to use the blocks? If writing your own code is important, you can use the [dvbs2-rx python code](https://github.com/igorauad/gr-dvbs2rx/blob/master/apps/dvbs2-rx) as a reference.

> But no constellation is archived after DVBS2 PL Sync.
I suspect that I have to set ACM/VCM mode to False but I do not know what values must entered at PLS filter entries.

Answering this question, note that the current version of gr-dvbs2rx does not support ACM/VCM fully. Only the PL Sync block supports ACM/VCM processing, and there the goal is to help with detecting the PLS settings (MODCOD, frame size and pilots) when these are not known a priori (e.g., trying to tune to an undocumented/unknown carrier). All blocks other than the PL Sync must be configured for a single MODCOD (CCM mode).

I understand you don't want ACM/VCM anyway, since you are trying to receive the QO100 beacon. For that, you can set the "ACM/VCM mode" option to False on the PL Sync block. 

As for the PLS filter, even when operating in CCM mode, you may want to accept multiple PLS values. If you know the pilot configuration (on or off), then you can work with a single PLS value. Otherwise, you may want to accept two possible PLS values, one for pilots off, the other for pilots on. See how they are defined here: https://github.com/igorauad/gr-dvbs2rx/blob/master/apps/dvbs2-rx#L725. And see this routine for setting the PLS filters: https://github.com/igorauad/gr-dvbs2rx/blob/master/python/dvbs2rx/params.py#L231.

The low and high filters are two 64-bit (u64) variables with bit masks for the 128 possible PLS values. The PLS has 7 bits, so it has 128 possibilities. Each 7-bit value has the MODCOD (5 bits) and the TYPE field (2 bits), and the latter indicates whether the PLFRAME has normal or short size and whether it contains PL pilots.

The PLSC (Reed-Muller encoded) of each incoming PLFRAME is decoded and the PL Sync block checks whether the incoming PLS value is one of the accepted ones, according to the PLS filters. See https://github.com/igorauad/gr-dvbs2rx/blob/master/lib/plsync_cc_impl.cc#L917. If you end up using the wrong filters, the PL Sync block will achieve frame lock but won't output the decoded XFECFRAMEs downstream for the subsequent blocks (LDPC/BCH decoders, etc). You can tell whether this is happening by checking the count of rejected frames. With `dvbs2-rx`, if you run with option `--log-stats`, you will see this count at the `frame_count.rejected` field in the periodic JSON-formatted logs.

Let me know if that is helpful at all and whether I can close the issue for now.

Thanks!

sv1bds, 2022-08-01:
I try this :
dvbs2-rx --source rtl --gui --sym-rate 1.5M --samp-rate 3M --modcod QPSK4/5 --rolloff 0.25 --pilots auto -f 741500000 --rtl-gain 20 -d 3
/usr/local/bin/dvbs2-rx:16: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.version import StrictVersion
log :info: Starting DVB-S2 Rx
log :warning: Warning: failed to XInitThreads()
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 3000000,178814 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
frame_sync :debug: --------------------------------------------------
frame_sync :debug: PLFRAME found
frame_sync :debug: Peak after: 1233396; Timing Metric: 32.010540; Locked: 0
frame_sync :debug: Sym: 1233396; SOF: +14.6 +0.3j; PLSC: +17.4 %+.1fj
plsync_cc :debug: SOF count: 1; Index: 1233306
freq_sync :debug: PLHEADER phase: 0.129828
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490
frame_sync :debug: --------------------------------------------------
frame_sync :debug: Peak after: 26004; Timing Metric: 31.689281; Locked: 0
frame_sync :debug: Sym: 26004; SOF: +14.7 +0.0j; PLSC: +16.9 %+.1fj
plsync_cc :debug: SOF count: 2; Index: 1259310
freq_sync :debug: PLHEADER phase: 0.567732
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490
frame_sync :debug: --------------------------------------------------
frame_sync :debug: Peak after: 51961; Timing Metric: 30.537136; Locked: 0
frame_sync :debug: Sym: 51961; SOF: +13.1 +0.2j; PLSC: +17.4 %+.1fj
plsync_cc :debug: SOF count: 3; Index: 1311271
freq_sync :debug: PLHEADER phase: 0.548318
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490
frame_sync :debug: --------------------------------------------------
frame_sync :debug: Peak after: 103828; Timing Metric: 30.980776; Locked: 0
frame_sync :debug: Sym: 103828; SOF: +14.0 +0.4j; PLSC: +17.0 %+.1fj
plsync_cc :debug: SOF count: 4; Index: 1415099
freq_sync :debug: PLHEADER phase: 0.42755
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490
frame_sync :debug: --------------------------------------------------
frame_sync :debug: Peak after: 51961; Timing Metric: 34.359901; Locked: 0
frame_sync :debug: Sym: 51961; SOF: +15.1 -0.2j; PLSC: +19.2 %+.1fj
plsync_cc :debug: SOF count: 5; Index: 1467060
freq_sync :debug: PLHEADER phase: 0.564757
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490
frame_sync :debug: --------------------------------------------------
[image]

It seems that I can not receive.
Any help please ?

igorauad, 2022-08-01:
Hi @sv1bds ,

That's good progress. I can see the timing metric is not that much strong yet. Ideally, the timing metric goes up to 57 when the signal has high SNR and the AGC performs its job properly. The frame sync algorithm will lock after only two consecutive timing metrics above 30 observed at the right (expected timing). That has not happened on your logs. I believe you would see that if the `Peak After` log were 32490. If this beacon signal is QPSK 4/5 with normal FECFRAME size and pilots off, then you have 361 slots of 90 symbols, hence an interval of 32490 symbols.

There are many things you can try. Perhaps you can message me on Telegram (username igorauad) for discussing in real-time if you wish. 

I would try the following:
- Using a higher gain on the RTL-SDR. From 20 to 40 dB gain is the range I typically use.
- Try with `--pilots off` if the pilots are really off in this carrier.
- Try with `--pl-acm-vcm -d3` if you are not sure about pilots and frame configuration.
- Try with the in-tree symbol sync algorithm using option `--sym-sync-impl in-tree`. That's a little more CPU-intensive but a bit more flexible in configurations. You can also play with the `--sym-sync-loop-bw` parameter, but let's leave that for later.

In any case, the signal looks pretty clear. I think you will get it working soon.

sv1bds, 2022-08-02:
Hi,
These are d 3 debuga data from a personal flowchart using Pluto for Rx.
I am unsure for short or normal FEC setting.
Can you determine the short/normal from these data?
I get a MER of about  10 but still can not get 4 groups of dots at constellation. 
Any idea ?
rame_sync :debug: --------------------------------------------------
frame_sync :debug: Peak after: 19069; Timing Metric: 59.164867; Locked: 0
frame_sync :debug: Sym: 19069; SOF: +25.7 +0.4j; PLSC: +33.5 %+.1fj
plsync_cc :debug: SOF count: 545; Index: 16862263
freq_sync :debug: PLHEADER phase: 1.8342
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490
frame_sync :debug: --------------------------------------------------
frame_sync :debug: Peak after: 20197; Timing Metric: 59.724411; Locked: 0
frame_sync :debug: Sym: 20197; SOF: +25.9 -1.0j; PLSC: +33.8 %+.1fj
plsync_cc :debug: SOF count: 546; Index: 16882460
freq_sync :debug: PLHEADER phase: 1.37593
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490
frame_sync :debug: --------------------------------------------------
frame_sync :debug: Peak after: 20559; Timing Metric: 62.262856; Locked: 0
frame_sync :debug: Sym: 20559; SOF: +27.9 -0.4j; PLSC: +34.3 %+.1fj
plsync_cc :debug: SOF count: 547; Index: 16903019
freq_sync :debug: PLHEADER phase: 1.49037
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490
frame_sync :debug: --------------------------------------------------
frame_sync :debug: Peak after: 11734; Timing Metric: 56.077709; Locked: 0
frame_sync :debug: Sym: 11734; SOF: +24.9 +1.1j; PLSC: +31.1 %+.1fj
plsync_cc :debug: SOF count: 548; Index: 16914753
freq_sync :debug: PLHEADER phase: 2.09109
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490
frame_sync :debug: --------------------------------------------------
frame_sync :debug: Peak after: 20203; Timing Metric: 60.772141; Locked: 0
frame_sync :debug: Sym: 20203; SOF: +27.1 +0.1j; PLSC: +33.6 %+.1fj
plsync_cc :debug: SOF count: 549; Index: 16934956
freq_sync :debug: PLHEADER phase: 1.23626
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490
frame_sync :debug: --------------------------------------------------
frame_sync :debug: Peak after: 19985; Timing Metric: 60.077423; Locked: 0
frame_sync :debug: Sym: 19985; SOF: +25.9 -0.2j; PLSC: +34.2 %+.1fj
plsync_cc :debug: SOF count: 550; Index: 16954941
freq_sync :debug: PLHEADER phase: 1.62958
plsc_decoder :debug: MODCOD:  8; Short FECFRAME: 0; Pilots: 0
plsc_decoder :debug: n_mod: 2; S: 360; PLFRAME length: 32490

sv1bds, 2022-08-02:
Hi again,
There is a bug in QO100 beacon and output is broken.
I have to verify first proper beacon operation before testing your software...
Thanks

igorauad, 2022-08-02:
Hi @sv1bds ,

You will only see the four constellation points after the frame sync stage, whose success is indicated by the `Locked: 1` frame_sync log. You haven't reached this state yet on the logs.

After frame synchronization, the other forms of synchronization kick in. The coarse/fine freq offset corrections and the phase correction. After that, the four constellation points become visible.

The timing metric is stronger on the latest logs but the interval between frame sync timing metric peaks is not consistent yet. Do you have any IQ data loss by any chance? If you are using an RTL-SDR, I'd recommend testing the RTL with `rtl_test -s 3000000`. Not all RTL-SDRs will be able to sample at 3 Msps without IQ loss. Not sure if that's your case.

sv1bds, 2022-08-02:
Hi,
Yes you are right! It loose data. It can receive to about 2400 max.

igorauad, 2022-08-22:
Hi, @sv1bds . It seems like I can close this issue for now. Let me know if you need any further help, and feel free to reopen the issue if necessary.

closed 2022-08-22 by igorauad