igorauad/gr-dvbs2rx #11: installation errors

sv1bds, 2022-06-23:
Hi, 
In a fresh 22.04 installation I have to add: sudo apt-get install libsndfile1-dev
Also
13/13 Testing: qa_symbol_sync_cc
13/13 Test: qa_symbol_sync_cc
Command: "/usr/bin/sh" "qa_symbol_sync_cc_test.sh"
Directory: /home/sv1bds/gr-dvbs2rx/build/python/dvbs2rx
"qa_symbol_sync_cc" start time: Jun 23 08:27 EEST
Output:
----------------------------------------------------------
/usr/lib/python3/dist-packages/gnuradio/gr/top_block.py:48: DeprecationWarning: setDaemon() is deprecated, set the daemon attribute instead
  self.setDaemon(1)
.F..
======================================================================
FAIL: test_open_loop (__main__.qa_plsync_cc)
Test open-loop mode
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/sv1bds/gr-dvbs2rx/python/dvbs2rx/qa_symbol_sync_cc.py", line 99, in test_open_loop
    self.assertListEqual(x_syms[1:], self.sink.data())
AssertionError: Lists differ: [(-1+0j), (3+0j), (-2+0j), (2+0j), (-2.5+0j), (1.5+0j)] != [(-1+0j), (3+0j), (-2+0j), (2+0j), (-2.5+0j)]

First list contains 1 additional elements.
First extra element 5:
(1.5+0j)

- [(-1+0j), (3+0j), (-2+0j), (2+0j), (-2.5+0j), (1.5+0j)]
?                                             ----------

+ [(-1+0j), (3+0j), (-2+0j), (2+0j), (-2.5+0j)]

----------------------------------------------------------------------

igorauad, 2022-07-05:
Hi @sv1bds ,

Thanks for opening this issue. 

> I have to add: sudo apt-get install libsndfile1-dev

That's right, you need to install `libsndfile1-dev`. That's actually a problem with the gnuradio-dev package, which I think should include `libsndfile1-dev` as a dependency to enable OOTs depending on gr-blocks.

If you are unsure about what to install, you can follow the Docker image recipe at: https://github.com/igorauad/gr-dvbs2rx/blob/master/Dockerfile. You can replicate the steps there on a fresh Ubuntu installation.

> AssertionError: Lists differ: [(-1+0j), (3+0j), (-2+0j), (2+0j), (-2.5+0j), (1.5+0j)] != [(-1+0j), (3+0j), (-2+0j), (2+0j), (-2.5+0j)]

Regarding this error on `qa_plsync_cc`, you are right, it does show up on Ubuntu 22.04. I was able to reproduce it. I investigated and concluded the problem was due to a change on `scipy.signal.upfirdn`, a function from SciPy used on this particular test case. Ubuntu 22.04 has a more recent version of SciPy (1.8), which includes the referred behavioral change and broke the test. I've pushed a fix to the master branch (commit cec3540). 

Let me know if you have more questions.

777arc, 2022-08-12:
Would be nice to add `sudo apt-get install libsndfile1-dev` to the install instructions, since the latest GR off the PPA doesn't include it, so most folks wont have it.

igorauad, 2022-08-22:
Hi, @777arc. Thanks for the suggestion. The installation instructions now mention `libsndfile1-dev` and the other build dependencies.

I'm going to close this issue for now. Please feel free to reopen or create another one.

closed 2022-08-22 by igorauad