igorauad/gr-dvbs2rx #12: dvbs2_tx_rx.grc example gives "plsync" error

pavja2, 2022-06-29:
For some reason I am unable to run the provided `examples/dvbs2_tx_rx.grc` flowgraph. Regardless of the `.ts` file I provide it, I always receive the error:

`thread_body_wrapper :error: ERROR thread[thread-per-block[16]: <block plsync_cc(8)>]: pmt_cdr: wrong_type 0.00299287`

Steps to replicate:
* Open `dvbs2_tx_rx.grc` in gnuradio-companion
* Change `in_file` to point to the sample file `test-3packets-04-05-06.ts` from https://tsduck.io/streams/?name=test-patterns
* Run flowgraph

The command line tools dvbs2tx and dvbs2rx work as expected. The example for `dvbs2_pl_sync.grc` gives the same error. Using a larger `.ts` file doesn't seem to fix anything as the flowgraph still crashses on the first error.

I've can also replicate this issue from grc within the provided Docker container:
`docker run -it --volume "/tmp/.X11-unix:/tmp/.X11-unix:rw" --env="DISPLAY" --env="QT_X11_NO_MITSHM=1" -v $HOME/:/home/<myhomedir>/ gr-dvbs2rx gnuradio-companion`

igorauad, 2022-06-29:
Hi @pavja2 , my bad. I know what caused this. I changed the rotator block recently to an in-tree version with slightly different format for a tag used to mark phase increment updates. I don't always use/test the example flow graphs, as I'm always using the main apps (`dvbs2-tx`/`dvbs2-rx`) instead, so I didn't catch this. Will push a fix.

igorauad, 2022-06-29:
I've just pushed f503755. Let me know if it works for you. It looks like it's working again.

pavja2, 2022-06-29:
Wow! Thanks for the crazy fast response and fix. I can confirm that it now works perfectly.

igorauad, 2022-06-29:
Awesome 👍 So I'll close the issue now. Feel free to open another one if you have more questions.

closed 2022-06-29 by igorauad