igorauad/gr-dvbs2rx #13: --gui option in dvbs2-rx application segmentation fault

muaddib1984, 2022-07-02:
Hi there, i was hoping to use your application to receive a blockstream satellite message, but when I run ```dvbs2-rx --gui``` I get the following error:
```
log :info: Starting DVB-S2 Rx
Traceback (most recent call last):
  File "/usr/local/bin/dvbs2-rx", line 1364, in <module>
    main()
  File "/usr/local/bin/dvbs2-rx", line 1321, in main
    tb = DVBS2RxTopBlock(options)
  File "/usr/local/bin/dvbs2-rx", line 130, in __init__
    self.setup_gui()
  File "/usr/local/bin/dvbs2-rx", line 435, in setup_gui
    self._add_gui_block(
  File "/usr/local/bin/dvbs2-rx", line 387, in _add_gui_block
    gui_obj_win = sip.wrapinstance(gui_obj.qwidget(), Qt.QWidget)
TypeError: wrapinstance() argument 2 must be sip.wrappertype, not sip.wrappertype
Segmentation fault (core dumped)
```
I can see [here](https://github.com/igorauad/gr-dvbs2rx/blob/f503755a586edcd7eda72658c86c4edd6206555c/apps/dvbs2-rx#L382-L383) that you've added accomodations for QT compatibility with GNURadio 3.9/3.10, but it doesn't seem to be work.

I am running Ubuntu 21.04 with GNURadio 3.10.2
Do you have any thoughts on how to fix this?

igorauad, 2022-07-05:
Hi @muaddib1984 ,

Thanks for reporting the issue. I was able to reproduce it on Ubuntu Jammy (22.04) and I've pushed a fix to the master branch See commit a4a05af. 

Let me know if that solves your problem.

muaddib1984, 2022-07-07:
@igorauad Works perfectly, thank you!
[image]

closed 2022-07-07 by muaddib1984

igorauad, 2022-07-07:
Awesome, @muaddib1984 ! Thanks for confirming.