igorauad/gr-dvbs2rx #15: Code refuses to link on Fedora 36

deeptho, 2022-07-30:
With the default compile flags, a linker error occurs on fedora 36 (linux, x86-64) because that code is compiled without -fPIC, which is necessary for shared libraries:
ldpc_decoder/libldpc_decoder_avx2.a(ldpc_decoder_avx2.cc.o): relocation R_X86_64_32 against hidden symbol '_ZN9ldpc_avx211LdpcDecoderE' can not be used when making a shared object

I succeeded in making it link and run by adding, in
lib/CMakeLists.txt:
+add_compile_options ( -fPIC )

A second such change has to be made in
in cpu_features/CMakeLists.txt but that is a git submodule.
So my workaround is probably not the best solution.

igorauad, 2022-08-19:
Hi @deeptho ,

I was not able to reproduce your problem on fc36. Could you send further instructions?

As far as I understand, the `-fPIC` flag comes from the `CMAKE_POSITION_INDEPENDENT_CODE` cmake option, which is set to `ON` by default for shared libraries. [See the cmake docs](https://cmake.org/cmake/help/latest/prop_tgt/POSITION_INDEPENDENT_CODE.html#prop_tgt:POSITION_INDEPENDENT_CODE).

I'm normally using Ubuntu for development, so I had to try on fc36. I used the following Dockerfile recipe to try a reproducible test.

```dockerfile
FROM fedora:36
RUN dnf install -y \
    fftw-devel \
    g++ \
    git \
    gmp-devel \
    gnuradio-devel \
    libsndfile-devel \
    python3-pybind11 \
    spdlog-devel
RUN git clone --recursive https://github.com/igorauad/gr-dvbs2rx.git && \
    cd gr-dvbs2rx/ && \
    mkdir build && \
    cd build/ && \
    cmake .. && \
    make -j4 && \
    make install
RUN cd gr-dvbs2rx/build/ && make test
```

If you save the above as a file called `fc36.docker` and run the command below, it should work:

```
docker build -t gr-dvbs2rx-fc36 -f fc36.docker .
```

deeptho, 2022-08-22:
I do not know about the docker command, but running 
the following command
cd /tmp
git clone --recursive https://github.com/igorauad/gr-dvbs2rx.git && \
    cd gr-dvbs2rx/ && \
    mkdir build && \
    cd build/ && \
    cmake .. && \
    make -j 8
results in:
/usr/bin/ld: ldpc_decoder/libldpc_decoder_avx2.a(ldpc_decoder_avx2.cc.o): relocation R_X86_64_32 against hidden symbol `_ZN9ldpc_avx211LdpcDecoderE' can not be used when making a shared object
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lib/CMakeFiles/gnuradio-dvbs2rx.dir/build.make:329: lib/libgnuradio-dvbs2rx.so.1.0.0.0] Error 1
make[1]: *** [CMakeFiles/Makefile2:453: lib/CMakeFiles/gnuradio-dvbs2rx.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

Afterwards,  It tried

sudo dnf install -y \
    fftw-devel \
    g++ \
    git \
    gmp-devel \
    gnuradio-devel \
    libsndfile-devel \
    python3-pybind11 \
    spdlog-devel
which only installed pythonr3-pybind11 
and then:

cd /tmp
rm -fr gr-dvbs2rx/

git clone --recursive https://github.com/igorauad/gr-dvbs2rx.git && \
    cd gr-dvbs2rx/ && \
    mkdir build && \
    cd build/ && \
    cmake .. && \
    make -j 8

This again produced the error.

make VERBOSE=1 showed the error below

/usr/lib64/ccache/c++ -fPIC  -march=native -Wall -O3 -DNDEBUG -shared -Wl,-soname,libgnuradio-dvbs2rx.so.1.0.0 -o libgnuradio-dvbs2rx.so.1.0.0.0 CMakeFiles/gnuradio-dvbs2rx.dir/bbdeheader_bb_impl.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/bbdescrambler_bb_impl.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/bch_decoder_bb_impl.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/ldpc_decoder_cb_impl.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/pi2_bpsk.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/pl_descrambler.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/pl_frame_sync.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/pl_freq_sync.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/pl_signaling.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/plsync_cc_impl.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/reed_muller.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/rotator_cc_impl.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/symbol_sync_cc_impl.cc.o CMakeFiles/gnuradio-dvbs2rx.dir/util.cc.o  ldpc_decoder/libldpc_decoder_avx2.a ldpc_decoder/libldpc_decoder_sse41.a ldpc_decoder/libldpc_decoder_generic.a ../cpu_features/libcpu_features.a /usr/lib64/libgnuradio-filter.so.3.10.1.0 /usr/lib64/libgnuradio-fft.so.3.10.1.0 /lib64/libfftw3f.so /lib64/libfftw3f_threads.so /usr/lib64/libgnuradio-blocks.so.3.10.1.0 /usr/lib64/libgnuradio-runtime.so.3.10.1.0 /usr/lib64/libgnuradio-pmt.so.3.10.1.0 /usr/lib64/libboost_program_options.so /usr/lib64/libboost_system.so /usr/lib64/libboost_regex.so /usr/lib64/libboost_thread.so /usr/lib64/libboost_chrono.so /usr/lib64/libboost_date_time.so /usr/lib64/libboost_atomic.so /usr/lib64/libspdlog.so.1.10.0 /usr/lib64/libfmt.so.8.1.1 -Wl,--as-needed /usr/lib64/libgmpxx.so /usr/lib64/libgmp.so -lrt /usr/lib64/libvolk.so.2.5.0 -ldl -lm /usr/lib64/libsndfile.so 
/usr/bin/ld: ldpc_decoder/libldpc_decoder_avx2.a(ldpc_decoder_avx2.cc.o): relocation R_X86_64_32 against hidden symbol `_ZN9ldpc_avx211LdpcDecoderE' can not be used when making a shared object
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lib/CMakeFiles/gnuradio-dvbs2rx.dir/build.make:329: lib/libgnuradio-dvbs2rx.so.1.0.0.0] Error 1
make[2]: Leaving directory '/tmp/gr-dvbs2rx/build'
make[1]: *** [CMakeFiles/Makefile2:453: lib/CMakeFiles/gnuradio-dvbs2rx.dir/all] Error 2
make[1]: Leaving directory '/tmp/gr-dvbs2rx/build'
make: *** [Makefile:146: all] Error 2

The PIC flag is set during link, but also needs to be set at compile time,

Third attempt, to check for possibly corrupt installed libraries or development files (unlikely because problem is in a compiled object:
sudo dnf reinstall -y \
    fftw-devel \
    g++ \
    git \
    gmp-devel \
    gnuradio-devel \
    libsndfile-devel \
    python3-pybind11 \
    spdlog-devel

cd /tmp
rm -fr gr-dvbs2rx/

git clone --recursive https://github.com/igorauad/gr-dvbs2rx.git && \
    cd gr-dvbs2rx/ && \
    mkdir build && \
    cd build/ && \
    cmake .. && \
    make -j 8

Same error.

Of course the tools installed in a docker image may be different and/or there may be something on my system left over from an old installation. I also tested my version 
of cmake: latest version. 

So I tried another installation in  a virtual machine. This was created from the official install uso, but some dev tools installed and some upgrades. Same error there after executing the commands you used (installation of prerequisities and building)

igorauad, 2022-08-22:
ok, @deeptho , I was able to reproduce the problem on a native fedora 36 (not a container).

I've just pushed a fix in e04e6fe. Let me know if that works for you.

deeptho, 2022-08-22:
@igorauad

This fix works. Nice software!

igorauad, 2022-08-22:
Awesome, @deeptho ! Thanks for confirming.

closed 2022-08-22 by igorauad