igorauad/gr-dvbs2rx #19: Hardware suggestion

sumanthraikar, 2023-03-09:
I decided to use the GNU flowgraph because I want data in and out of the signal processing modules in real-time and/or offline decoding of DVBS2 signals from satellites in India using an SDR (I want to connect to INSAT-4A or 4B satellite with a freely available broadcast channel in freq range 10-12Ghz). The local DTH operates in Ku band so I am planning to use the antenna and LNB (universal LNB's sold by local vendors in kit) should the universal LNB be a concern?
In order to give the downconverted signal to the SDR should I be concerned about using bias-T or SWM power inserter with respect to overall performance or quality of signal. Are they both the same?
Also, assuming SWM, should I be concerned about input voltage readings of it? If yes then how to find to desired power to feed the LNB?

igorauad, 2023-03-10:
Hi @sumanthraikar ,

>  should the universal LNB be a concern?

It is a concern when trying to receive a signal within the Ku high band from 11.7 to 12.75 GHz. The reason is that the Universal LNB has two LO frequencies, 9750 MHz and 10600 MHz, which are switched by a 22 kHz tone supplied to the LNB. By default, the Universal LNB operates with the 9750 MHz LO, which allows the reception of the Ku low band (10.7 to 11.7 GHz) with the typical range of L-band tuners. If you supply a 22 kHz tone, it switches to the 10600 MHz LO and can receive the Ku high band.

The problem is that it may be challenging to supply a 22 kHz tone with an SDR. So, it's easier to pick an LNB that gives you the frequency you want. But if you want free-to-air TV channels, I believe most will be in the Ku low band, so you should be fine. For instance, check [INSAT-4B on lyngsat](https://www.lyngsat.com/Insat-4B-and-G-Sat-15.html), and you will see the Ku-band frequencies are < 11.7 GHz. Nevertheless, if you are using an RTL-SDR, check the tuner range of your model (e.g., 24 MHz to 1766 MHz with the R820T/T2) to make sure you can get the signal of interest.

> should I be concerned about using bias-T or SWM power inserter with respect to overall performance or quality of signal?

That shouldn't be a problem. That will only affect the DC voltage supplied to the LNB but shouldn't affect the RF or downconverted signal (at least, not to my knowledge). However, keep in mind that linear-polarized Ku-band LNBs typically support both horizontal and vertical polarization, and the supplied DC voltage determines the polarization. A voltage of around 13 V puts the LNB in vertical polarization and 18 V in horizontal polarization. Worst case, if you can't control the DC voltage, you only need to rotate the LNB (change the polarity angle) to compensate for that.

You can find plenty of instructions about antenna/LNB pointing and LNB hardware choices in the documentation I sent you. Here are some relevant sections:
- https://blockstream.github.io/satellite/doc/antenna-pointing.html#mount-the-antenna
- https://blockstream.github.io/satellite/doc/hardware.html#universal-lnb
- https://blockstream.github.io/satellite/doc/hardware.html#setup-specific-components

sumanthraikar, 2023-03-13:
From this [page](https://www.lyngsat.com/tvchannels/in/DD-National.html). I can see GSAT-15 broadcasting this channel at 11630V (is V for vertical polarisation?) and in SR FEC column it says 3000, 3/5 (i am wondering what is this?) If 3/5 is the code-rate, can I still use the GNU radio flowgraph to get a live feed of this channel? How to change the code rate in flowgraph (I am seeing coderate dependent on modulation order which is not clear with the usual code rate definition of k/n where k is encoding input size and n is encoding out size)?

igorauad, 2023-03-13:
Hi @sumanthraikar ,

- Yes, V is for Vertical polarization.
- 3000 is the baud rate (or symbol rate), which you can configure on `dvbs2-rx` with option `--sym-rate`.
- 3/5 is the code rate, as you said. The MODCOD is 8PSK 3/5 in this case (see the 8PSK modulation noted on lyngsat). Hence, you need to run `dvbs2-rx` with option `--modcod 8psk3/5`.

See the docs about parameters https://igorauad.github.io/gr-dvbs2rx/docs/usage.html#experimenting-with-parameters.

igorauad, 2023-03-13:
By the way, 30e6 symbols per second is a 36 MHz carrier with the typical 0.2 roll-off factor. Note you would need a relatively powerful/expensive SDR interface and CPU to process that in real time. Even if you can sample that fast, you may find that the CPU won't be able to handle the FEC decoders (BCH and LDPC) in real time.

I'd recommend starting with a narrow carrier if you can find one. The Blockstream Satellite service has 1.2 MHz carriers, carrying 1 Mbaud signals, which you can sample with an RTL-SDR at 2 Msamples/sec. There is coverage in India via the Telstar 18V C-band beam. See https://blockstream.com/satellite/#satellite_network-coverage. 

You could also try the QO-100 DVB-S2 beacon, which I think is 1500 kbaud QPSK 4/5. See #10 .

LamaBleu, 2023-03-16:
Hi sorry for interfering !  
Confirming I'm able to receive QO100 beacon using RTLSDR with  following settings :  
```
dvbs2-rx --source rtl -f 741.443e6 --samp-rate 2.4e6 -s 1500e3 --gui --sink file --out-file /tmp/test1.ts --modcod QPSK4/5  -p off  --rtl-gain 45  
```
- Since I'm using a small 60cm dish, I'm just on the limit and had to deal a lot with gain and samplerate (got nothing using default SR @3MSps)  
- Frequency is 741.5 MHz +/- LNB offset (57kHz here)  
Thank you @igorauad !  

[image]  
[image]

sumanthraikar, 2023-03-17:
Can I use a 21V LNB[LNB](Directv 21 Volt Power Inserter For SWM Integrated LNB (2 pack) https://amzn.eu/d/huGTeun)  power supply by directTV to get blockstream satellite data? I can’t find 13 and 18V variants at the moment.
Also does the blockstream satellite broadcasts video or text information?

igorauad, 2023-03-17:
> Can I use a 21V LNB[LNB](Directv 21 Volt Power Inserter For SWM Integrated LNB (2 pack) https://amzn.eu/d/huGTeun) power supply by directTV to get blockstream satellite data? I can’t find 13 and 18V variants at the moment.
> Also does the blockstream satellite broadcasts video or text information?

Yes, you can definitely use a 21V SWM power inserter. That is the model most people use with an SDR setup. You can always rotate the LNB to compensate for the polarity shift.

The Blockstream Satellite signal broadcasts UDP/IP packets (data) carrying Bitcoin blockchain information. There is no video in the stream, but you can still somewhat validate the data by observing packets on `tcpdump` or something like that.

sumanthraikar, 2023-03-22:
Can you suggest an antenna, (some kind of an up-converter) and the required hardware in order to transmit dvbs2 out in real-time?