igorauad/gr-dvbs2rx #20: AttributeError

wangjiliang1983, 2023-04-18:
Hi,

I opened dvbs2_tx_rx.grc simulation file to run in gnuradio. But I got the error in terminal.

> Generating: '/home/demo/workarea/gr-dvbs2rx/examples/dvbs2_tx_rx.py
>
>Executing: /usr/bin/python3 -u /home/demo/workarea/gr-dvbs2rx/examples/dvbs2_tx_rx.py
>
>Traceback (most recent call last):
> File "/home/demo/workarea/gr-dvbs2rx/examples/dvbs2_tx_rx.py", line 751, in <module>
>   main()
>  File "/home/demo/workarea/gr-dvbs2rx/examples/dvbs2_tx_rx.py", line 729, in main
>    tb = top_block_cls(agc_gain=options.agc_gain, agc_rate=options.agc_rate, agc_ref=options.agc_ref, debug=options.debug, frame_size=options.frame_size, freq_offset=options.freq_offset, gold_code=options.gold_code, in_file=options.in_file, modcod=options.modcod, pl_freq_est_period=options.pl_freq_est_period, rolloff=options.rolloff, rrc_delay=options.rrc_delay, rrc_nfilts=options.rrc_nfilts, snr=options.snr, sps=options.sps, sym_rate=options.sym_rate, sym_sync_damping=options.sym_sync_damping, sym_sync_loop_bw=options.sym_sync_loop_bw)
>  File "/home/demo/workarea/gr-dvbs2rx/examples/dvbs2_tx_rx.py", line 343, in __init__
>    self.dvbs2rx_rx_hier_0 = dvbs2rx_rx_hier(
>  File "/home/demo/.grc_gnuradio/dvbs2rx_rx_hier.py", line 89, in __init__
>    *dvbs2rx.params.translate('DVB-S2',
>  File "/usr/local/lib/python3.8/dist-packages/gnuradio/dvbs2rx/params.py", line 148, in translate
>    standard, frame_size, constellation = _adjust_case(standard, frame_size,
>  File "/usr/local/lib/python3.8/dist-packages/gnuradio/dvbs2rx/params.py", line 19, in _adjust_case
>    constellation = constellation.upper()
>AttributeError: 'int' object has no attribute 'upper'
>
>>>> Done (return code 1)

I'm using gnuradio 3.10.5.1, ubuntu 20.04 and python 3.8.10

wangjiliang1983, 2023-04-18:
I tried command line tools with example 1 &2. they works well.
But, I hope to use dvbs2rx with gnuradio and USRPB210, So we can tune the RF frequency on the fly.

wangjiliang1983, 2023-04-19:
Hi @igorauad,
Thanks for help. The dvbs2rx_tx_rx.grc works now.
I can also got example9  running on two USRPs.
And I got this log after running dvbs2rx:
>[INFO] [B200] Asking for clock rate 32.000000 MHz... 
>[INFO] [B200] Actually got clock rate 32.000000 MHz.
>log :info: Lock=False
>Ousrp_source :error: In the last 1639 ms, 1 overflows occurred.
>bbdeheader_bb :warning: Baseband header invalid (dfl > kbch - 80).
>log :info: Lock=True; SNR=16.43; FECFRAMEs=64; FER=9.1e-01; TS Packets=42; PER=1.4e-01
>log :info: Lock=True; SNR=31.48; FECFRAMEs=288; FER=2.0e-01; TS Packets=1453; PER=4.1e-03
>log :info: Lock=True; SNR=27.21; FECFRAMEs=544; FER=1.1e-01; TS Packets=3065; PER=2.0e-03
>log :info: Lock=True; SNR=26.76; FECFRAMEs=768; FER=7.6e-02; TS Packets=4476; PER=1.3e-03
>log :info: Lock=True; SNR=26.60; FECFRAMEs=1024; FER=5.7e-02; TS Packets=6088; PER=9.9e-04
>log :info: Lock=True; SNR=26.70; FECFRAMEs=1248; FER=4.6e-02; TS Packets=7499; PER=8.0e-04

Is it possible to recalculate FER and PER so the error frame and error packets at the every beginning can be ignored?

wangjiliang1983, 2023-04-19:
I used command line tools with --gui on 2x  USRP B210s for some time. Both tx and rx work fine if I don't tune gain or Center frequency.
However, if I change tx_gain,from low to high or high to low and when the lock status changes, there is a huge chance that the RX crashes with Segmentation Fault (core dumped). And there is often an overflow in usrp_source.

I checked the CPU usage: there is no core over 70% usage.

BTW, if dvb2rx_tx_rx.grc is used instead of command line, There are more overflows happening.

> Ousrp_source :error: In the last 3198 ms, 1 overflows occurred.
> Ousrp_source :error: In the last 1162 ms, 1 overflows occurred.
> OOusrp_source :error: In the last 1188 ms, 2 overflows occurred.
> Ousrp_source :error: In the last 1217 ms, 2 overflows occurred.
> Obbdeheader_bb :warning: Baseband header invalid (dfl > kbch - 80).
> OOusrp_source :error: In the last 1246 ms, 2 overflows occurred.
> OOusrp_source :error: In the last 1223 ms, 2 overflows occurred.
> Obbdeheader_bb :warning: Baseband header invalid (dfl > kbch - 80).
> Ousrp_source :error: In the last 1176 ms, 2 overflows occurred.
> OOusrp_source :error: In the last 1194 ms, 2 overflows occurred.
> Obbdeheader_bb :warning: Baseband header invalid (dfl > kbch - 80).
> usrp_source :error: In the last 1165 ms, 2 overflows occurred.
> OOusrp_source :error: In the last 1646 ms, 1 overflows occurred.
> usrp_source :error: In the last 2184 ms, 1 overflows occurred.
> OOOusrp_source :error: In the last 1197 ms, 2 overflows occurred.
> bbdeheader_bb :warning: Baseband header invalid (dfl not a multiple of 8).
> OOusrp_source :error: In the last 1241 ms, 2 overflows occurred.
> OOusrp_source :error: In the last 1195 ms, 2 overflows occurred.
> 
> 
I thought it may be an performance issue. So,I tried to recompile with BENCHMARK_CPU and BENCHMARK_FEC turned ON to do some benchmark. But the cmake complained about AFF3CT_DIR not defined after I installed both aff3ct and libbenchmark. I checked the aff3ct runs well.

igorauad, 2023-04-19:
Good progress, @wangjiliang1983 

> Is it possible to recalculate FER and PER so the error frame and error packets at the every beginning can be ignored?

Yes, it is possible. However, I've been planning to improve this through another route. Instead of ignoring the initial errors, I would like to improve the performance during the initial transient. 

This behavior comes mainly from the PL Sync block, which locks the frame timing (finds where frame frames start) quickly, but takes a few dozen frames to synchronize well in frequency. The parameter `--pl-freq-est-period` controls how many frames are accumulated before outputting the first coarse frequency offset estimate. Until then, the performance is significantly less reliable. On the other hand, when considering the goal is to demodulate/decode data, it doesn't hurt to at least try processing these initial frames, so I currently let the decoders attempt to decode them, which is why you see the initial errors. We could:

1) Prevent the processing of frames until a coarse frequency offset estimate is obtained. The advantage would be less initial frame/packet errors. The disadvantage would be that those initial frames would be dropped completely.
2) Provide a mechanism for a faster (although noisier) coarse frequency offset estimation during the initial transient. Instead of waiting until 30 frames are accumulated, we can update the estimate per frame until 30 are accumulated. After that, we continue with the normal operation.

I'm more inclined to go with alternative 2, which has been in my backlog for upcoming work.

> However, if I change tx_gain,from low to high or high to low and when the lock status changes, there is a huge chance that the RX crashes with Segmentation Fault (core dumped).

I'd be interested in investigating this. I have an idea of what could be causing the problem. For a quick test, could you run the Rx side with `--sym-sync-impl in-tree`. See if the problem persists after that. Although, you may already be running the in-tree symbol sync implementation, depending on the sampling frequency you are using. Could you share the full command you are using on Tx and Rx?

> And there is often an overflow in usrp_source.

These overflows are not necessarily due to CPU. They could be caused by the interface with the USRP (USB or Ethernet). With the B210, I'd start by making sure the USRP is connected to a USB 3.0 port, not USB 2.0. With Ethernet USRPs, usually, you would want to connect it by a direct cable if possible. You can [benchmark your USRP](https://kb.ettus.com/Verifying_the_Operation_of_the_USRP_Using_UHD_and_GNU_Radio#Benchmarking_your_system) to confirm the achievable Tx/Rx rates.

> I thought it may be an performance issue. So,I tried to recompile with BENCHMARK_CPU and BENCHMARK_FEC turned ON to do some benchmark. But the cmake complained about AFF3CT_DIR not defined after I installed both aff3ct and libbenchmark. I checked the aff3ct runs well.

Those benchmarking features are more for development. I use them to decide whether a new decoder implementation is faster, so that the code does not have a regression. But that won't help you much. 

The [build option](https://github.com/igorauad/gr-dvbs2rx/blob/master/docs/installation.md#build-options) that matters for your use case is the `NATIVE_OPTIMIZATIONS`, but it is already on by default. Leave it on so that you can get the maximum performance.

wangjiliang1983, 2023-04-19:
Hi @igorauad ,

Here is the command I used:
For Tx side
> dvbs2-tx \
>     --source file \
>     --in-file example.ts \
>     --in-repeat \
>     --sink usrp \
>     --usrp-args "serial=30AC29E" \
>     --usrp-subdev A:A \
>     --usrp-clock-source internal \
>     --usrp-sync pc_clock \
>     --usrp-gain 30 \
>     --modcod QPSK1/4 \
>     --frame short \
>     --rolloff 0.35 \
>     --sym-rate 2000000 \
>     --pilots \
>     --gui 

For Rx Side
> dvbs2-rx \
>     --log \
>     --source usrp \
>     --usrp-args "serial=311CDA6" \
>     --usrp-subdev A:A \
>     --usrp-clock-source internal \
>     --usrp-sync pc_clock \
>     --usrp-gain 30 \
>     --sink file \
>     --out-file /dev/null \
>     --modcod QPSK1/4 \
>     --frame short \
>     --rolloff 0.35 \
>     --sym-rate 2000000 \
>     --gui

I'll try `--sym-sync-impl in-tree` next time.

I did benchmarked USRP using benchmark_rate command coming with UHD. There is no problem setting sampling rate to 56Msps.

wangjiliang1983, 2023-04-20:
Hi @igorauad ,

I played around command line dvbs2-tx dvbs2-rx with simulation mode. I changed the sps to 4 and sym_rate to 2M. There is a big chance the RX will crash with segmentation fault. Here is the command I use and the output log

> demo@demo-OptiPlex-7000:~/Desktop$ dvbs2-tx \
> >     --source file \
> >     --in-file example.ts \
> >     --in-repeat \
> >     --modcod QPSK1/4 \
> >     --frame short \
> >     --sps 4 \
> >     --rolloff 0.35 \
> >     --sym-rate 2000000 \
> >     --pilots \
> >     --gui | \
> > dvbs2-rx \
> >     --log \
> >     --sink file \
> >     --out-file /dev/null \
> >     --modcod QPSK1/4 \
> >     --frame short \
> >     --sps 4 \
> >     --rolloff 0.35 \
> >     --sym-rate 2000000 \
> >     --gui
> log :info: Starting DVB-S2 Rx
> log :info: Lock=False
> log :info: Lock=True; SNR=26.98; FECFRAMEs=672; FER=0.0e+00; TS Packets=1336; PER=0.0e+00
> log :info: Lock=True; SNR=27.01; FECFRAMEs=1216; FER=0.0e+00; TS Packets=2419; PER=0.0e+00
> Segmentation fault (core dumped)

Here is another try:

demo@demo-OptiPlex-7000:~/Desktop$ dvbs2-tx \
>     --source file \
>     --in-file example.ts \
>     --in-repeat \
>     --modcod QPSK1/4 \
>     --frame short \
>     --sps 4 \
>     --rolloff 0.35 \
>     --sym-rate 2000000 \
>     --pilots \
>     --gui \
>     --snr 10 | \
> dvbs2-rx \
>     --log \
>     --sink file \
>     --out-file /dev/null \
>     --modcod QPSK1/4 \
>     --frame short \
>     --sps 4 \
>     --rolloff 0.35 \
>     --sym-rate 2000000 \
>     --gui
log :info: Starting DVB-S2 Rx
log :info: Lock=False
Segmentation fault (core dumped)

>

wangjiliang1983, 2023-04-20:
One more thing:
 RX did not crash when using in-tree symbol sync.

igorauad, 2023-04-20:
@wangjiliang1983 Thanks for reporting the segfault problem. I could reproduce it and pushed a fix in abb4082. There are still other possibilities for segfaulting the symbol sync block, one of which is covered in another branch (not on master yet), specifically 7595837. However, the latter is still under test, as it may penalize performance a bit (and the symbol sync block is supposed to be as fast as possible).

Please let me know if you still experience segmentation faults.

closed 2023-04-22 by wangjiliang1983

AndresRRR15, 2025-04-10:
I'm wanting to send a video stream using this 

Tx:

ffmpeg -f v4l2 -i /dev/video0 -c:v libx264 -preset ultrafast -tune zerolatency -b:v 2.3M -r 30 -s 640x480 -f mpegts -muxrate 2500000 -mpegts_flags system_b -mpegts_copyts 1 udp://127.0.0.1:12347

Rx

ffplay -i /tmp/received.ts -vf "fps=30,scale=640:480" -fflags nobuffer -analyzeduration 10000000 -probesize 10000000

and I can send the text file for the tx and rx 

[Rx.txt](https://github.com/user-attachments/files/19693914/Rx.txt)
[Tx.txt](https://github.com/user-attachments/files/19693913/Tx.txt)