igorauad/gr-dvbs2rx #31: Syndrome calculation

daniestevez, 2023-11-02:
(More of a comment than an actual issue). I'm looking at the new BCH decoder code to get an idea of where the performance improvements come from.

I saw that the following trick is used to evaluate syndromes: If $r(x)$ is the received polynomial (which in DVB-S2 has very large degree, on the order of 10000), instead of evaluating $r(x)$ at the $2t$ elements $\alpha, \alpha^2, \ldots, \alpha^{2t} \in GF(2^{16})$ to obtain the syndromes, we observe that $r(\alpha^j) = q_j(\alpha^j)$, where $q_j(x)$ is the quotient obtained by dividing $r(x)$ by the minimal polynomial of $\alpha^j$ in $GF(2)[x]$. It is also observed that some of the $\alpha^j$, $j = 1,\ldots,2t$ have the same minimal polynomial (i.e., they are conjugates), so the calculation of $q_j(x)$ can be re-used for several $\alpha^j$.

However, my understanding is that within the set of the $\alpha^j$, $j = 1, \ldots, 2t$, the conjugates come in pairs. In other words, there are $t$ distinct $q_j(x)$ polynomials to compute. These are the quotients obtained when dividing $r(x)$ by the first $t$ polynomials $g_k(x)$ in either Table 6a or Table 6b in the [DVB-S2 standard](https://www.etsi.org/deliver/etsi_en/302300_302399/30230701/01.04.01_60/en_30230701v010401p.pdf).

An alternative idea is to divide $r(x)$ instead by the product $g(x) = g_1(x) g_2(x) \cdots g_t(x)$. This polynomial $g(x)$ is the code generator polynomial and has degree $16 t$. Since $g(\alpha^j) = 0$ for $j = 1,\ldots,2t$, if we denote the remainder of the division of $r(x)$ by $g(x)$ as $p(x)$, then the syndromes can be calculated by evaluating $p(x)$ at $\alpha^j$ for $j = 1,\ldots, 2t$.

Comparing both methods, the current method requires $t$ polynomial divisions on $GF(2)[x]$ of a polynomial of very large degree by a polynomial of degree 16, plus $2t$ evaluations of different polynomials of degree 15 in $GF(2)[x]$ at elements of $GF(2^{16})$. The method I propose requires one polynomial division on $GF(2)[x]$ of a polynomial of very large degree by a polynomial of degree $16t$, plus $2t$ evaluations of a single polynomial of degree $16t - 1$ at elements of $GF(2^{16})$. Since these elements are consecutive powers of $\alpha$, they can be calculated with a variant of the algorithm in `gf2m_poly<T>::search_roots_in_exp_range` (which is basically Chien's search).

I was wondering if you have given some thought to the performance of these two approaches or have done some benchmarks. To me it is not immediately obvious which one would be faster.

igorauad, 2023-11-04:
Brilliant idea, @daniestevez ! Thanks a lot for suggesting it. I had totally missed this possibility.

My reference for the implementation was Lin and Costello's Error Control Coding book. I started the investigation with a [Python-based implementation](https://github.com/igorauad/bch). Then, I moved it to C++ and worked on optimizations. However, I missed your suggested approach after following the recipe described in Section 6.2 of the book.

You are definitely right. It is better to have a single polynomial division and evaluate a longer remainder polynomial (degree up to $m t$) than to compute $t$ polynomial divisions and evaluate shorter remainder polynomials (degree up to $m-1$).

There were some constraints in the implementation preventing me from doing that before. I have explained it in 37a89c2. Also, I have given a go to your suggestion in 4499c1b. I still need to clean up the implementation, but the results are exciting. Here is the new benchmark:

```
# * Parameters: 
#    ** Frame errors   = 1000
#    ** Max frames     = 1000
#    ** Noise seed     = 0
#    ** Info. bits (K) = 38688
#    ** Frame size (N) = 38880
#    ** Err. Corr. (t) = 12
#    ** Code rate  (R) = 0.995062
#    ** SNR min   (dB) = 0
#    ** SNR max   (dB) = 10
#    ** SNR step  (dB) = 1
#
# ---------------------||------------------------------------------------------||---------------------
#  Signal Noise Ratio  ||   Bit Error Rate (BER) and Frame Error Rate (FER)    ||  Global throughput  
#         (SNR)        ||                                                      ||  and elapsed time   
# ---------------------||------------------------------------------------------||---------------------
# ----------|----------||----------|----------|----------|----------|----------||----------|----------
#     Es/N0 |    Eb/N0 ||      FRA |       BE |       FE |      BER |      FER ||  SIM_THR |    ET/RT 
#      (dB) |     (dB) ||          |          |          |          |          ||   (Mb/s) | (hhmmss) 
# ----------|----------||----------|----------|----------|----------|----------||----------|----------
      -0.02 |     0.00 ||     1000 |  3063231 |     1000 | 7.92e-02 | 1.00e+00 ||   40.109 | 00h00'00   
       0.98 |     1.00 ||     1000 |  2195881 |     1000 | 5.68e-02 | 1.00e+00 ||   41.485 | 00h00'00   
       1.98 |     2.00 ||     1000 |  1462750 |     1000 | 3.78e-02 | 1.00e+00 ||   40.025 | 00h00'00   
       2.98 |     3.00 ||     1000 |   896832 |     1000 | 2.32e-02 | 1.00e+00 ||   40.279 | 00h00'00   
       3.98 |     4.00 ||     1000 |   492725 |     1000 | 1.27e-02 | 1.00e+00 ||   40.299 | 00h00'00   
       4.98 |     5.00 ||     1000 |   236306 |     1000 | 6.11e-03 | 1.00e+00 ||   41.496 | 00h00'00   
       5.98 |     6.00 ||     1000 |    94727 |     1000 | 2.45e-03 | 1.00e+00 ||   39.292 | 00h00'00   
       6.98 |     7.00 ||     1000 |    31210 |     1000 | 8.07e-04 | 1.00e+00 ||   40.682 | 00h00'00   
       7.98 |     8.00 ||     1000 |      865 |       60 | 2.24e-05 | 6.00e-02 ||   47.322 | 00h00'00   
       8.98 |     9.00 ||     1000 |        0 |        0 | 2.58e-08 | 1.00e-03 ||   58.182 | 00h00'00   
  ```

The previous results are by the end of the [FEC benchmark page](https://github.com/igorauad/gr-dvbs2rx/tree/master/bench/fec#bch-decoder). The approach you suggested is gaining another 4 to 7 Mbps of throughput!

daniestevez, 2023-11-06:
Great!

Here are some more ideas for further performance improvements.

First, regarding benchmarking, I would approach it somewhat differently. The throughput of the BCH decoder depends almost exclusively on the number of bit errors, with maximum throughput for correct frames, and worst throughput for `t` errors and uncorrectable frames. This is because the error locator polynomial has degree `e`, where `e` is the number of errors, so the operations required to find its roots using Chien's search is proportional to `e`. So I think that rather than doing an AFF3CT-style benchmark, it is better to benchmark for each number of bit errors separately, and look at the resulting curve. Also, the BCH decoder is algebraic. It is either correctly implemented (and corrects all error patterns up to `t` errors) or it has a bug, so looking at BER and FER curves is not so important as for an LDPC decoder.

Now, for concrete ideas:

* Once you have computed the remainder of the message polynomial divided by the code generator polynomial (this is the first stage I suggested), you can check if the remainder if zero. If so, the message is correct and you can finish. There is no need to compute the syndromes on that polynomial. Since when used for DVB-S2 the output of the LDPC decoder is correct for most frames, it is good to optimize the case when there are 0 errors.
* The previous BCH decoder implementation had special cases to find the roots of the error locator polynomial for degrees 1 and 2 (corresponding to messages with 1 or 2 bit errors). Again, these cases occur often when using this for DVB-S2, so it's good to optimize them. The case of degree 1 is trivial. The polynomial is `ax + b = 0`, so we can just solve for `x` immediately. This is way faster than Chien's search. The case of degree 2 uses some clever math, but also gives the solutions for `x` as a formula (using a lookup table) rather than as trial and error, so it's fast. You can look at `LocationFinder` and `ArtinSchreier` in the older code.

igorauad, 2023-11-07:
Thanks again, @daniestevez ! 

I agree with your comments about the benchmarking approach. I use the current implementation while being aware that the lower throughput at low SNRs is natural because there are more errors to correct. I agree it would be nice to visualize the throughput while varying the number of errors per codeword.

By the way, when the noise is high, and there are likely more than $t$ errors per codeword, there is a condition that could help here: [bch.cc#L306](https://github.com/igorauad/gr-dvbs2rx/blob/bch_syndrome/lib/bch.cc#L306). If I understand correctly, having degree $> t$ on the error-location polynomial is theoretically possible. However, in practice, I find that this condition is hardly hit, so it doesn't save CPU. Any thoughts?

Regarding the other suggestions:

1. I'm now skipping the syndrome computation when $r(x) % g(x)$ is zero. I had a condition before to skip the computation of the error-location polynomial and numbers when the syndrome was an all-zeros vector. That is preserved, and the computation is now skipped when the syndrome returns an empty vector in the updated implementation. See 5415348.
2. I have implemented the suggested special case for finding the root of a unit-degree error-location polynomial. See cb5ef45. 
 
The performance improved again with the two updates above, but it wasn't as significant as the first update (dividing by g(x) instead of minimal polynomials). Naturally, it made a more significant difference for high SNR since the changes apply when there are no errors or a single error.

```
# ---------------------||------------------------------------------------------||---------------------
#  Signal Noise Ratio  ||   Bit Error Rate (BER) and Frame Error Rate (FER)    ||  Global throughput  
#         (SNR)        ||                                                      ||  and elapsed time   
# ---------------------||------------------------------------------------------||---------------------
# ----------|----------||----------|----------|----------|----------|----------||----------|----------
#     Es/N0 |    Eb/N0 ||      FRA |       BE |       FE |      BER |      FER ||  SIM_THR |    ET/RT 
#      (dB) |     (dB) ||          |          |          |          |          ||   (Mb/s) | (hhmmss) 
# ----------|----------||----------|----------|----------|----------|----------||----------|----------
      -0.02 |     0.00 ||     1000 |  3063231 |     1000 | 7.92e-02 | 1.00e+00 ||   41.239 | 00h00'00   
       0.98 |     1.00 ||     1000 |  2195881 |     1000 | 5.68e-02 | 1.00e+00 ||   41.709 | 00h00'00   
       1.98 |     2.00 ||     1000 |  1462750 |     1000 | 3.78e-02 | 1.00e+00 ||   42.254 | 00h00'00   
       2.98 |     3.00 ||     1000 |   896832 |     1000 | 2.32e-02 | 1.00e+00 ||   41.747 | 00h00'00   
       3.98 |     4.00 ||     1000 |   492725 |     1000 | 1.27e-02 | 1.00e+00 ||   41.875 | 00h00'00   
       4.98 |     5.00 ||     1000 |   236306 |     1000 | 6.11e-03 | 1.00e+00 ||   41.517 | 00h00'00   
       5.98 |     6.00 ||     1000 |    94727 |     1000 | 2.45e-03 | 1.00e+00 ||   41.703 | 00h00'00   
       6.98 |     7.00 ||     1000 |    31210 |     1000 | 8.07e-04 | 1.00e+00 ||   41.230 | 00h00'00   
       7.98 |     8.00 ||     1000 |      865 |       60 | 2.24e-05 | 6.00e-02 ||   48.171 | 00h00'00   
       8.98 |     9.00 ||     1000 |        0 |        0 | 2.58e-08 | 1.00e-03 ||   62.307 | 00h00'00   
```

The next step is to figure out how to implement the special case for a quadratic error-location polynomial. If you have any references to point, I'd appreciate that!

Cheers

daniestevez, 2023-11-08:
> If I understand correctly, having degree on the error-location polynomial is theoretically possible.

I don't remember the details well, and maybe it depends on the algorithm you use to compute the error-locator polynomial (Berlekamp-Massey, etc.). I think that the algorithm is guaranteed to give you a polynomial of degree `t` or less (basically because that's the number of unknown that the algorithm is solving) but it can fail. In the case of a BCH decoder, if it fails, decoding stops with an uncorrectable codeword.

> I had a condition before to skip the computation of the error-location polynomial and numbers when the syndrome was an all-zeros vector. That is preserved, and the computation is now skipped when the syndrome returns an empty vector in the updated implementation.

If the remainder of the division of the message polynomial by the code generator polynomial is not zero, then there is at least on syndrome that is nonzero. So it doesn't pay off to check whether all the syndromes are zero if you follow this approach.

> The next step is to figure out how to implement the special case for a quadratic error-location polynomial. If you have any references to point, I'd appreciate that!

For reference, the code is [here](https://github.com/igorauad/gr-dvbs2rx/blob/5e5e52d9c8d0974923bdd47ad114d26b9322f955/lib/reed_solomon_error_correction.hh#L83).

The math works as follows. Given a quadratic polynomial $ax^2 + bx + c \in GF(2^{16})[x]$ with $a \neq 0$, if it is reducible, its roots $x_0$ and $x_1$ should satisfy $x_0 x_1 = c/a$, and $x_0 + x_1 = b/a$. Denoting $R=ax_0/b$, we get $x_1 = b(R+1)/a$, and so $c/a = x_0 x_1 = b^2R(R+1)/a^2$. This means that $R$ should satisfy the equation $R(R+1) = ca/b^2$.

This equation can be solved by using a table. We compute $R(R+1)$ for each $R \in GF(2^{16})$ and store the value $R$ in the entry with index $R(R+1)$. If we assume that $c \neq 0$ (otherwise the roots of the quadratic polynomial are trivially seen to be $0$ and $b/a$, and I'm not sure if this case was handled correctly in the code), then $R \neq 0$. Therefore, in the table we can use the value $0$ for those entries $\alpha$ for which the equation $R(R+1) = \alpha$ does not have a solution.

Coming back to solving the equation, we compute $ca/b^2$, use this as index on the table to get $R$, and then compute the roots as $x_0 = bR/a$ and $x_1 = b(R+1)/a$.

igorauad, 2023-11-19:
Very interesting, @daniestevez . Thanks for the clarification! The math makes total sense. I have pushed an implementation in 20aaccb.

Regarding the case of $c = 0$, I believe the LUT approach to finding the roots of the error-location polynomial would work, but not the inversion to obtain the *error-location numbers*. In this case, $c a / b^2$ is zero, so $R (R+1)$ is equivalently zero. With the implementation in 20aaccb, we have `LUT[0] = 0`, so the looked-up value of $R$ would be zero. Finally, plugging $R=0$ into the expressions for $x_0$ and $x_1$, we would obtain $x_0 = 0$ and $x_1 = b/a$, as expected.

However, as you know, the problem is that we cannot have zero roots, given these are not invertible. The approach I adopted in 20aaccb is to return a single error-location number $1/x_1$ in this case. Also, if $\sigma(x) = ac^2$, I return an empty vector of error-location numbers, as all roots are zero. Nevertheless, I don't know if the error-location polynomial generated by the simplified Berlekamp iterative procedure (the one I'm using) would ever have one of these forms. Let me know if you have any comments.

As for the performance, this change enhances the throughput by another Mbps or so at the 9 dB Eb/N0 level presented in the previous benchmarks. The other lower Eb/N0 levels did not change much, given they were mostly affected by more than two errors. I've pushed the updated results in 45d76c9.

> If the remainder of the division of the message polynomial by the code generator polynomial is not zero, then there is at least on syndrome that is nonzero. So it doesn't pay off to check whether all the syndromes are zero if you follow this approach.

You are right. I removed the verification in bf923b2. Now, I check directly if the syndrome vector is empty or not. It will be empty when the remainder in the division of the message polynomial by the code generator polynomial is zero.

daniestevez, 2023-11-19:
For the case $c = 0$, the roots are easily seen to be $x = 0$ and $x = b/a$. But as you point out, all the roots of the error locator polynomial must be invertible, so we can stop early with a decoding failure in this case.

Additionally (and this is something I glossed over in my previous message), the roots of the error-locator polynomial must be all distinct. This forces $b \neq 0$, because if $b = 0$, then the error locator polynomial is $a x^2 + c$. If there is an $x_0$ such that $x_0^2 = c/a$, then $a(x + x_0)^2 = ax^2 + c$ (because in a field of characteristic 2 we have $(x+y)^2 = x^2 + y^2$ for all $x,y$), so we see that the root of the error-locator polynomial is double, and we can also finish early with a decoding failure.

In fact, the previous implementation of this algorithm contains the checks $b = 0$ and $c = 0$, finishing immediately in any of these.
https://github.com/igorauad/gr-dvbs2rx/blob/5e5e52d9c8d0974923bdd47ad114d26b9322f955/lib/reed_solomon_error_correction.hh#L84

igorauad, 2023-11-19:
I see, @daniestevez . I'm curious whether the root $b/a$ is useful when the other root is $0$ (in the case of $c=0$). If $b/a$ could save one bit, it may be worth returning it, as it could save one TS packet that would otherwise fail CRC. This is what I'm assuming here: https://github.com/igorauad/gr-dvbs2rx/blob/bch_syndrome/lib/bch.cc#L367

But if $b/a$ (or error-location number $a/b$) doesn't help when $c=0$, then I'd agree it makes sense to return early with decoding failure. Thoughts?

daniestevez, 2023-11-19:
I don't think your reasoning is correct. The assumption for BCH decoding is that the codeword has $e$ errors or less. If the assumption doesn't hold, then decoding must fail unless you want to do an "exotic" form of BCH decoding that can sometimes correct more than $e$ errors (which probably comes at the cost of not being able to correct some patterns of $e$ errors or less). Under this assumption, we have $c \neq 0$. If $c = 0$ this implies that there are more than $e$ errors and decoding must fail.

igorauad, 2023-11-19:
I see, @daniestevez . So, I've updated the implementation. It is now in c6fc8af. The updated section checking b==0 and c==0 is here: https://github.com/igorauad/gr-dvbs2rx/blob/bch_syndrome/lib/bch.cc#L352

daniestevez, 2023-11-21:
Taken a look at c6fc8af and it looks good to me.

igorauad, 2023-11-21:
Thanks, @daniestevez !

closed 2023-11-21 by igorauad