igorauad/gr-dvbs2rx #34: bladeRF does not work

DuchIvan, 2024-01-30:
Hi,
I am trying to run:
`dvbs2-tx --source file --in-file ~/Videos/1min_timer_mpegts.ts --sink bladeRF --freq 1316.9e6 --sym-rate 1e6 --usrp-args "serial=9e4" `
and my error is:
```
Traceback (most recent call last):
  File "/usr/local/bin/dvbs2-tx", line 1055, in <module>
    main()
  File "/usr/local/bin/dvbs2-tx", line 1027, in main
    tb = dvbs2_tx(options)
  File "/usr/local/bin/dvbs2-tx", line 178, in __init__
    sink_block = self.connect_sink()
  File "/usr/local/bin/dvbs2-tx", line 576, in connect_sink
    sink = self.setup_blade_rf_sink()
  File "/usr/local/bin/dvbs2-tx", line 461, in setup_blade_rf_sink
    import bladeRF
ModuleNotFoundError: No module named 'bladeRF'

I have bladeRF library compiled and installed from sources. And next commands works:
```
python3
import bladerf
```
Please help me understand what is "import bladeRF" and how can I get it?

andreamassetti, 2025-02-19:
I have the same issue, were you able to solve this?

DuchIvan, 2025-06-06:
No, unfortunately.