igorauad/gr-dvbs2rx #42: Example 13 results in tsdump: cannot detect TS file format

cafeclimber, 2024-08-19:
Trying to run Example 13 with the following command:

```
cat example.ts | dvbs2-tx --out-fd 3 3>&1 1>&2 | dvbs2-rx --out-fd 3 3>&1 1>&2 | tsp --realtime --buffer-size-mb 1 -O play
```

results in the following output:

```
log :info: Starting DVB-S2 Rx
tsdump: cannot detect TS file format
```

and then the application terminates. All of the other examples I've tried have worked, and running the following

```
cat example.ts | tsp --realtime --buffer-size-mb 1 -O play
```

does result in video playback.

Wondering if I'm missing something? Possible bug?

sethp, 2025-06-25:
I ran into something very similar; it turned out to be a "feature" of my shell (zsh): [Multios]( https://zsh.sourceforge.io/Doc/Release/Redirection.html#Multios). In short: if you're using zsh (i.e. `echo $SHELL` includes `zsh`), you should try closing fd 1 before redirecting fd 3 into the pipe. In other words: 

```zsh
<example.ts dvbs2-tx --out-fd 3 3>&1 1>&2 | dvbs2-rx --out-fd 3 3>&1 1>&- 1>&2 | tsp --realtime --buffer-size-mb 1 -O play         
```

Note the addition of the `1>&-` in between the `1>&2` and `3>&1`. There are other options (`... dvbs2-rx --out-fd 3 3> >(tsp ... play`, or disabling the MULTIOS setting), but the explicit close has the benefit of working equally well under `bash` as well (I didn't try any other shells).

To explain a little more what's going on there: zsh behaves differently than bash. When you redirect to the same file descriptor twice (as in: the implicit stdout for fd1 and then the `3>&1`), by default zsh spins up a subprocess that duplicates the data to _both_, whereas bash will _replace_ the fd in the new processes' table. 

What this means for us is that zsh is "helpfully" copying not just the received MPEG transport stream to tsduck, but also all of the logging that's also going out over stderr (fd 2). We can see that by looking at the first few bytes of output:

```
<example.ts dvbs2-tx --out-fd 3 3>&1 1>&2 | dvbs2-rx --out-fd 3 3>&1 1>&2 | hexdump -C | head
log :info: Starting DVB-S2 Rx
00000000  6c 6f 67 20 3a 69 6e 66  6f 3a 20 53 74 61 72 74  |log :info: Start|
00000010  69 6e 67 20 44 56 42 2d  53 32 20 52 78 0a 47 00  |ing DVB-S2 Rx.G.|
00000020  31 14 57 78 f9 9b 61 3e  18 c4 b1 d4 93 b5 ea 3f  |1.Wx..a>.......?|
```

The printable bytes corresponding to "log :info: Start" do not, in fact, signal the valid start of an MPEG TS format, so tsduck gets confused and gives up. Closing the file descriptor in between explicitly tells zsh not to do this, and we see just the `47` ["magic" byte](https://en.wikipedia.org/wiki/List_of_file_signatures) that signals the beginning of the transport stream:   

```
$ <example.ts dvbs2-tx --pilots --out-fd 3 3>&1 1>&2 | dvbs2-rx --out-fd 3 3>&1 1>&- 1>&2 | hexdump -C | head
log :info: Starting DVB-S2 Rx
00000000  47 00 31 14 57 78 f9 9b  61 3e 18 c4 b1 d4 93 b5  |G.1.Wx..a>......|
00000010  ea 3f 39 5c 31 a3 7f 3b  c0 03 26 b3 06 43 3c 3e  |.?9\1..;..&..C<>|
```

I also found this discussion very helpful in understanding what ZSH is doing: https://unix.stackexchange.com/questions/673145/zsh-understanding-sequence-of-redirections-and-pipe-with-pointers

cafeclimber, 2025-06-26:
Omg I'll have to look at this! I AM using zsh. Thank you for the detailed response!!

cafeclimber, 2025-07-30:
Can confirm running in Bash (and the command you gave) work!