igorauad/gr-dvbs2rx #43: bladeRF module not found

JonathanB98, 2024-09-12:
Hi, 
I am trying to run dvbs2-rx with a bladeRF source like this:
`dvbs2-rx --source bladeRF --in-real-time --freq 433e6 --sym-rate 1e6 --gui`

Unfortunately, the bladeRF Python-module can not be found:
```log :info: Starting DVB-S2 Rx
Traceback (most recent call last):
  File "/usr/local/bin/dvbs2-rx", line 1630, in <module>
    main()
  File "/usr/local/bin/dvbs2-rx", line 1586, in main
    tb = DVBS2RxTopBlock(options)
         ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/dvbs2-rx", line 148, in __init__
    source_block = self.connect_source()
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/dvbs2-rx", line 745, in connect_source
    source = self.setup_blade_rf_source()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/dvbs2-rx", line 618, in setup_blade_rf_source
    import bladeRF
ModuleNotFoundError: No module named 'bladeRF
```

I installed the libbladerf-Python-Bindings from here:
[Nuand/bladeRF](https://github.com/Nuand/bladeRF/tree/master/host/libraries/libbladeRF_bindings/python)
and I can import the bladerf (**but not the bladeRF**) Module in Python:
```
$ python3
Python 3.12.3 (main, Jul 31 2024, 17:43:48) [GCC 13.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import bladeRF
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'bladeRF'
>>> import bladerf
>>>
```
The bladerf-Module can also be imported in dvbs2-rx, when I change the `import bladeRF`-line to `import bladerf as bladeRF`
This fixed the ModuleNotFoundError, but apparently the Module is missing the `source`-Attribute:
```
log :info: Starting DVB-S2 Rx
Traceback (most recent call last):
  File "/usr/local/bin/dvbs2-rx", line 1630, in <module>
    main()
  File "/usr/local/bin/dvbs2-rx", line 1586, in main
    tb = DVBS2RxTopBlock(options)
         ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/dvbs2-rx", line 148, in __init__
    source_block = self.connect_source()
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/dvbs2-rx", line 745, in connect_source
    source = self.setup_blade_rf_source()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/dvbs2-rx", line 620, in setup_blade_rf_source
    self.blade_rf_source = blade_rf_source = bladeRF.source(
                                             ^^^^^^^^^^^^^^
AttributeError: module 'bladerf' has no attribute 'source'
```
I suspect I have a version of the bladeRF module installed that is not compatible with dvbs2-rx or is there a whole other project for the library apart from the one in Nuand/bladeRF ?
Could someone tell me where I can get a compatible bladeRF-Library

wucke13, 2024-09-28:
I have the exact same behavior on my side.

wucke13, 2024-09-29:
@JonathanB98 after adding https://github.com/Nuand/gr-bladeRF to my gnuradio setup, rx with the bladeRF micro worked for me.

JonathanB98, 2024-09-30:
Thank you for the hint, I got it working with that

closed 2024-09-30 by JonathanB98